drlukeparry/pyslm #4: Hatch Order Issues

aacitelli, 2021-04-14:
Hello, 

I've been investigating using your library over the course of the last few weeks and am very impressed at what you've put together. 

However, I am a little confused by the output and hope you could clarify some things. I am slicing a pretty basic part and doing output like so: 

```python
# Create/wipe folder
    if not os.path.exists("LayerFiles"):
        os.makedirs("LayerFiles")
    else:
        for f in glob.glob("LayerFiles/*"):
            os.remove(f)

# Generate new output
for i in tqdm(range(len(layers)), desc="Generating Plots"):
    fig, ax = plt.subplots()
    PowderBedFusion.outputtools.plot(
        layers[i], plot3D=False, plotOrderLine=False, plotHatches=True, plotContours=True, handle=(fig, ax))
    fig.savefig("LayerFiles/Layer{}.png".format(i), bbox_inches='tight')
    plt.cla()
    plt.close(fig)
```

Whenever I use the basic hatcher, I get the following for a layer:
[image]

I would expect the colors to range from red to purple (or similar) across the entire part in a smooth progression. This seems to be the case for the top-right of the part, but this seems to fall apart drastically starting at the stripe just left of the maximum diagonal, and I can't really find any rhyme or reason as to why this occurs. 

Along a related vein, I have been experimenting with the implementation of island scanning that comes with the library. It gives me output like the following:
[image]

While this doesn't look too bad, it seems like the top-right of the bottom few islands would be burned before the bottom-left of the respective islands; the color increment just looks like that would occur. That's not really a criminal issue, but leads up to something else I tried. 

I tried to modify the implementation of island scanning as follows to implement a random island order instead:

```python
# Went from this...
for i in np.arange(0, numIslands):
            for j in np.arange(0, numIslands):

# ---------------------------
# To this...
# Generate random order
island_order = np.empty((numIslands ** 2, 2), dtype=int)
idx = 0
for i in np.arange(0, numIslands):
    for j in np.arange(0, numIslands):
        island_order[idx] = np.array([i, j])
        idx += 1
np.random.shuffle(island_order)

# Iterate through islands and gen hatches
for island_coords in island_order:
    i, j = island_coords[0], island_coords[1]
```

This seems to be all well and good, and runs without crashing. However, I'm confused, because the output looks the same as your implementation of it: 
[image]

I've double-checked that the modified code is being run, that the values of i/j are set correctly, and that the output is regenerated from the modified code. Ideally the "heatmap" progression would jump from island to island at random, but unfortunately it seems like that's just top to bottom. 

Whether that's related to the first issue or not, I'm unsure. I'm hoping you can comment on if this is an issue with the library itself, the way you interface with `matplotlib`, an issue with your codebase in general, or an issue with how I've hooked into `pyslm`. 

Thank you for your work and for your input!

drlukeparry, 2021-04-15:
Hi,

The order should be maintained across the part and in your case scan diagonally in one direction. To further visualise the scan vector order -  pass `plotOrderLine` as True, which shows the order of scanning via the midpoint of the hatch vectors.  If the `Hatcher` class is not working in the example, then there is clearly an underlying issue

Did you install the library from PyPi or clone the repo and use the library directly? I notice there is namespace change - did you use setup.py to compile the Python module you've adapted?

Until the first issue is resolved, I wouldn't expect that the random island scanning will work. In theory, your implementation should work correctly, as the vectors (including its ordering) within each island are added incrementally.

If you are happy to share a more complete script, I can perhaps understand if it is an error within the library or the installation and advise from there. 

All the best,
Luke

aacitelli, 2021-04-16:
Recloning and rebuilding the library was successful.

While a coworker _did_ do some namespace changes to make the library closer to a previous architecture we've been working with, I doubt that was the issue, as I suspect that would have just errored out at runtime due to import 

I suspect the issue was the second of your points; in the process of my coworker shifting over some stuff to what we were more familiar with, he ended up deleting some files that were pretty important - namely, that `setup.py` file, which I believe indicates the module wasn't being recompiled, or something similar. I'm not an expert on Python's exact compilation structure (particularly with respect to modules), and it's fixed, so it's good enough for me. 

The randomized version of island scanning is functional, if you're curious: 
[image]

I've gone ahead and essentially rolled back to using the vanilla version of the library. I'm not sure why we made any changes in the first place, but it's whatever, and we should be good now. Thank you for your help! I'll raise another issue if I run into anything else.

closed 2021-04-16 by aacitelli