drlukeparry/pyslm #5: Debug Output in Release Version

aacitelli, 2021-04-16:
Hello, 

While testing a fresh install of the library, there seems to be a lot of what equates to debug output (ignore the "Processing Layers"; that's just tqdm):
[image]

Is this intentional, or is there a good reason for this? I'd think output like this isn't useful to the end user and would be better off deleted or optionally enabled. It's messing with my tqdm progress bars and I wanted to double-check if there is a specific reason why this is still around in the library.

aacitelli, 2021-04-16:
I've opened a PR that (hopefully) addresses the issue: #6

drlukeparry, 2021-04-17:
Yes, I am still become  learning to becoming more accustomed to formalizing the releases better and removing redundant debug information. I am not a fan of the python logging library. 

I will go through the pull request. Some information such as updating the geometry cache information, I would like to keep verbose to quickly identify issues. However, during the slicing/hatching, I think this debug output is safe to now remove. 

I will try and pull this into a maintenance release. 

Many thanks for your contribution,

Luke

aacitelli, 2021-04-17:
The pull request just comments out essentially all the output I could find. It's likely not the cleanest solution, as I agree some output from the library can be very useful in many use cases. 

The cleanest way to do it, in my opinion (which is far from an expert one), would be some combination of the following:
- Tracking a boolean somewhere that toggles whether you want output from the library. This could be an optionally specified attribute in the `Part` class to control output for geometry-related options, and an optionally specified attribute in the `Hatcher` class (or similar) to control hatching-related output, so it's up to the end user whether they want the output. 
- Verbosity levels. For instance, fatal, error, warning, info, and debug, in declining order of importance. 

Using tqdm is a very solid way to visualize the project part, and output happening in the middle doesn't always mesh cleanly with that, so for my intended use case, the best solution would be one that allows me to stop all output that isn't a fatal error (warnings are also likely fine).

aacitelli, 2021-05-21:
Seems to be fixed on dev. Nice!

closed 2021-05-21 by aacitelli