drlukeparry/pyslm #10: Numba usage?

aacitelli, 2021-06-22:
I'm unsure if it's already implemented, but have you considered implementing parts of the library in [Numba](http://numba.pydata.org/)? It's is a library works in conjunction with Numpy that essentially implements JIT compilation in Python, speeding up loops and simple functions. The downside is that you have to write code in a style similar to C, but you can essentially selectively enable the library to only run on certain code blocks, which is done pretty simply via decorators.

drlukeparry, 2021-06-25:
Hi, 

For now, I haven't considered any specific lower-level optimisations as much of the time I have is on developing a reference-able accessible library that can be modified, rather than focus on a production grade slicer at this stage.

From the small time spent profiling, much of the underlying time spent is either across the slicing (via Trimesh) or the hatching operation performed by ClipperLib, which can  trivially made parallel across layers. Other operations matrix style calculations are done via numpy. 

For numba, it could be used as an optional dependency but I am unsure at this stage where use of the JIT optimisation would provide  a significant performance benefit?

aacitelli, 2021-07-01:
Was just curious on your thoughts as to if it would tie into the library; as most computation is done in dependency libraries I don't think there is much feasible usage here and your time is likely spent improving the library in other ways. Thanks!

closed 2021-07-01 by aacitelli