drlukeparry/pyslm #11: Error Installing PythonSLM from PyPi

aacitelli, 2021-07-06:
I have a bit of an interesting environment. I am trying to bundle all the requirements for `pyslm` into a portable Python installation, as I'm essentially building a UI that wraps around my organization's infrastructure for interfacing with PySLM. I was able to get an embeddable version of Python working with pip by following [this guide](https://www.christhoung.com/2018/07/15/embedded-python-windows/). I was then able to install 90% of the project dependencies to this embeddable version.

However, pip is running into the following error when it gets to the `triangle` package: 
```
ERROR: Command errored out with exit status 1:
   command: 'C:\Users\Kal\Documents\cdme-scangen-ui\python\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Kal\\AppData\\Local\\Temp\\pip-install-4oqlhgdx\\triangle_6607d6f5e9b84b9b9d04622fc8215807\\setup.py'"'"'; __file__='"'"'C:\\Users\\Kal\\AppData\\Local\\Temp\\pip-install-4oqlhgdx\\triangle_6607d6f5e9b84b9b9d04622fc8215807\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\Kal\AppData\Local\Temp\pip-wheel-020a81mv'
       cwd: C:\Users\Kal\AppData\Local\Temp\pip-install-4oqlhgdx\triangle_6607d6f5e9b84b9b9d04622fc8215807\
  Complete output (115 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.9
  creating build\lib.win-amd64-3.9\triangle
  copying triangle\core1_run.py -> build\lib.win-amd64-3.9\triangle
  copying triangle\data.py -> build\lib.win-amd64-3.9\triangle
  copying triangle\plot.py -> build\lib.win-amd64-3.9\triangle
  copying triangle\tri.py -> build\lib.win-amd64-3.9\triangle
  copying triangle\version.py -> build\lib.win-amd64-3.9\triangle
  copying triangle\__init__.py -> build\lib.win-amd64-3.9\triangle
  creating build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\A.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\bbox.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\box.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\box.2.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\box.3.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\box.4.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\diamond_02_00009.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\diamond_02_00009.1.v.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\diamond_02_00009.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\dots.1.v.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\dots.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex.2.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex2.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex2.2.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex3.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex3.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\ell.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\face.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\greenland.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\la.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\spiral.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\spiral.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\spiral.q.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\spiral.r.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\square_circle_hole.1.node -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\A.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\bbox.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\box.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\box.2.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\box.3.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\box.4.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\diamond_02_00009.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex.2.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex2.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex2.2.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\double_hex3.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\ell.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\face.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\greenland.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\la.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\spiral.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\spiral.q.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\spiral.r.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\square_circle_hole.1.ele -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\A.1.poly -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\A.poly -> build\lib.win-amd64-3.9\triangle\data
  copying triangle\data\box.1.poly -> build\lib.win-amd64-3.9\triangle\data
[... 35 lines cut ...]
  c/triangle.c(3734): warning C4311: 'type cast': pointer truncation from 'subseg *' to 'unsigned long'
  c/triangle.c(3739): warning C4311: 'type cast': pointer truncation from 'subseg *' to 'unsigned long'
  c/triangle.c(3744): warning C4311: 'type cast': pointer truncation from 'subseg *' to 'unsigned long'
  c/triangle.c(3780): warning C4311: 'type cast': pointer truncation from 'subseg *' to 'unsigned long'
  c/triangle.c(3785): warning C4311: 'type cast': pointer truncation from 'subseg *' to 'unsigned long'
  c/triangle.c(3792): warning C4311: 'type cast': pointer truncation from 'subseg *' to 'unsigned long'
  c/triangle.c(3801): warning C4311: 'type cast': pointer truncation from 'vertex' to 'unsigned long'
  c/triangle.c(3808): warning C4311: 'type cast': pointer truncation from 'vertex' to 'unsigned long'
  c/triangle.c(3815): warning C4311: 'type cast': pointer truncation from 'triangle *' to 'unsigned long'
  c/triangle.c(3822): warning C4311: 'type cast': pointer truncation from 'triangle *' to 'unsigned long'
  c/triangle.c(3831): warning C4311: 'type cast': pointer truncation from 'vertex' to 'unsigned long'
  c/triangle.c(3838): warning C4311: 'type cast': pointer truncation from 'vertex' to 'unsigned long'
  c/triangle.c(6966): warning C4244: '=': conversion from 'double' to 'int', possible loss of data
  c/triangle.c(12465): warning C4996: 'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
  C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DVOID=void -DREAL=double -DNO_TIMER=1 -DTRILIBRARY=1 -DANSI_DECLARATORS=1 -Ic -IC:\Users\Kal\Documents\cdme-scangen-ui\python\include -IC:\Users\Kal\Documents\cdme-scangen-ui\python\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tctriangle/core.c /Fobuild\temp.win-amd64-3.9\Release\triangle/core.obj
  core.c
  triangle/core.c(4): fatal error C1083: Cannot open include file: 'Python.h': No such file or directory
[... cut ...]
The error appears both when I'm trying to install prerequisites for a version of pyslm built from source (our organization has a custom version with minor differences that shouldn't be relevant) and when simply trying to use the `PythonSLM` version from PyPi. This process works correctly when I am not using the embeddable version - that is, I've gotten the library working in general, just not when I'm trying to package it with a UI application built in Electron that essentially launches the Python script as a child process. Do you have any ideas for how I may go about fixing this?

drlukeparry, 2021-07-06:
The error is related to the triangle library. Given it is looking for the Python source code, you would need to check that your Python environment is recognised - for this I can suggest that you look into Python embedded further.  I get the impression that the embedded version of python does not include the source headers as part of the distribution which is why the warning appears but it is new to me.

The more simple solution to your problem is that the triangle library is not currently needed yet in version 0.3. This potentially will be used for the support generation module in future, which is used by Trimesh - although I have an alternate function for this that may remove the dependency. However, for your purposes I think it can be safely removed without affecting the core hatching functionality.

aacitelli, 2021-07-08:
Removing `triangle` from my `requirements.txt` resulted in the error disappearing and I am now able to successfully package the library as a dependency alongside an Electron application. I've documented this issue so we can revisit it at a later point if we need support for future pyslm triangle-dependent features, but this is a fine solution for us for the next little while. Thank you!

closed 2021-07-08 by aacitelli