drlukeparry/pyslm #14: Facing error while running the code

adi18997, 2021-12-08:
zsh: command not found: import
zsh: command not found: import
zsh: command not found: import
from: can't read /var/mail/pyslm
zsh: number expected
zsh: number expected

Found when I ruined example after installing PythonSLM 
% pip install PythonSLM
Requirement already satisfied: PythonSLM in /opt/anaconda3/lib/python3.9/site-packages (0.4.2)
Requirement already satisfied: triangle in /opt/anaconda3/lib/python3.9/site-packages (from PythonSLM) (20200424)
Requirement already satisfied: numpy in /opt/anaconda3/lib/python3.9/site-packages (from PythonSLM) (1.20.3)
Requirement already satisfied: setuptools in /opt/anaconda3/lib/python3.9/site-packages (from PythonSLM) (58.0.4)
Requirement already satisfied: shapely in /opt/anaconda3/lib/python3.9/site-packages (from PythonSLM) (1.8.0)
Requirement already satisfied: trimesh in /opt/anaconda3/lib/python3.9/site-packages (from PythonSLM) (3.9.35)
Requirement already satisfied: cython in /opt/anaconda3/lib/python3.9/site-packages (from PythonSLM) (0.29.24)

Please help me out

drlukeparry, 2021-12-08:
Can I check you are using Linux/Mac OSX? 

Also, if you are using linux, are you installing this from a command prompt (BASH)? I am not sure how the Python setup scripts work with the zsh shell.

adi18997, 2021-12-08:
I am using MAC

adi18997, 2021-12-08:
If possible can we exchange our cellphone number so we can chat over there.
I am in trouble I have to submit a project on LPBF and I just go 4 days.

adi18997, 2021-12-08:
I am using MAC OS and I am doing it in Jupyternotebook.

closed 2021-12-21 by drlukeparry