drlukeparry/pyslm #16: LinearSort erroring out

aacitelli, 2022-02-21:
Hopefully this screenshot of my debugger demonstrates the error well enough:
[image]

The error comes from this line of sorting.py: https://github.com/drlukeparry/pyslm/blob/90a999f131eab20253f8396bc0e4250e11143832/pyslm/hatching/sorting.py#L108

The error is complaining that I'm taking the dot product of a 2-long array (essentially a unit vector representative of the hatch angle) and a X-long array of midpoints, where X is the number of vectors. My knowledge of linear algebra and numpy is sketchy at best, but isn't this expecting two same-length arrays? Is a dot product the proper way to handle this? 

I am running this with a pretty standard setup. I don't anticipate a specific test case will help much, as I think the error is apparent. I unfortunately just don't have enough background knowledge to fix it.

drlukeparry, 2022-02-21:
It's a  small issue, that is related to to this function being  un-documented. In short, hatches should have a numpy shape (n,2,2) so use `.reshape(-1,2,2)`.  I need to make this more consistent in future and also have built-in checks for shapes. Numpy automatically can 'broadcast' the vectors onto each other. 

The maths behind this sorting approach is simpler than it appears.

An orthogonal vector is constructed against the hatch-direction. This creates effectively an axis to project the hatch vectors on to. By taking the scalar or inner-product between the hatch vectors and this axis, the midpoint or any point of the vector are projected onto this axis. The hatches are then sorted using the projected distance along the axis.

aacitelli, 2022-02-21:
Well, it doesn't rectify the other issue I currently have up, but I figured out this one. At least in my code (which I don't *think* has any relevant changes), the `hatchAngle` attribute of the sort was never being set, so it was defaulting to zero degrees and thus essentially sorting them by the x-value of their midpoints. I fixed by adding this block early on in the hatch() method, which should set it once per layer:

```python
if self.hatchSortMethod:
    self.hatchSortMethod.hatchAngle = self.hatchAngle
```

closed 2022-02-21 by aacitelli