drlukeparry/pyslm #19: Error when try to replay the example: no module named'pyslm.visualise'

TomyTiger, 2022-03-25:
import pyslm
import pyslm.visualise
Traceback (most recent call last):
  File "<pyshell#5>", line 1, in <module>
    import pyslm.visualise
ModuleNotFoundError: No module named 'pyslm.visualise'

Can you please help?

TomyTiger, 2022-04-06:
I am using Python 3.10. Using IDLE to do the test.

drlukeparry, 2022-04-06:
Hi TomyTiger,

I confirmed the behavior in some of the PySLM examples. The examples and PySLM has been correctly updated to import pyslm.visualise correctly (commit [d21a3159cd2dfee0d995692bff3b409ba11efe17](https://github.com/drlukeparry/pyslm/commit/d21a3159cd2dfee0d995692bff3b409ba11efe17)), which may resolve the issue in the upcoming release (0.5)

closed 2022-09-06 by drlukeparry