drlukeparry/pyslm #22: Using the pyslm.hatching.*Hatcher with shapely object as input

jkur, 2022-09-06:
Hi,
i'm trying to use some of the pySLM methods to do some basic hatching for robot tool path generation (there are not so many hatching implementation out there).

The idea is to create a shapely.Polygon oder shapely.LineString from a set of arbitrary points (x,y).
Then i like to feed this polygon into one of the Hatchers from pySLM.

The examples use getVectorSlice() for generating input for the .hatch() method.

Can i generate the input for hatching from shapely Objects?
Or would it be possible to generate input directly based on some list of points that describe a polygon?

The whole point is to generate paths for some laser cladding application, where the surface boundaries are extracted from some CAD document via FreeCAD.

drlukeparry, 2022-09-06:
HI, 

Sounds an interesting application. 

That is fine and definitely possible.  As you suggested you can directly export the path segments as a series of coordinates from the shapely objects.  There's an internal function for extracting segment/path coordinates from shapely polygons into a format (list of coordinates) used internally by the clipper library in PySLM. This function iterates through the paths within a Shapely polygon list and extracts the coordinates into a compatible list of coordinates. 

```
 def path2DToPathList shapes) -> List[np.ndarray]:

    paths = []

    for poly in shapes:
        coords = np.array(poly.exterior.coords)
        paths.append(coords)

        for path in poly.interiors:
            coords = np.array(path.coords)
            paths.append(coords)

    return paths

coordList = path2DToPathList(shapelyPaths)
```

From here you create a Hatcher object (or sub-class) and use the  hatching functionality directly.

```
myHatcher = hatching.Hatcher()
layer = myHatcher.hatch(coordList)
```

jkur, 2022-09-08:
Sorry, but the problem is probably not with the hatching code from PySLM but somewhere in my code. With certain geometries everything looks actually quite nice. 

[image]
The tool path is projected to the XY-Plane for now. Therefore is looks a bit off. 

I think i need to dig a bit further.

drlukeparry, 2022-09-09:
That looks a nice result already within FreeCAD and integrating it with PySLM.

Likely, you have to be careful with the projection of coordinates into the OpenCascade wire (path) structure used in FreeCAD and ensure that they full connected and in the correct orientation (winding order, for the contour paths. If in doubt, extract the polygons, hatch and plot with `pyslm.visualise.plotPolygon`, or `pyslm.visualise.plot()`

The default Hatcher assumes that there is one offset for the outer contour, but this behavior can be changed in the future, if there are occurance of issues.

Let me know if you find any further issues.

drlukeparry, 2023-01-03:
No further questions, so marked as resolved.

closed 2023-01-03 by drlukeparry