drlukeparry/pyslm #23: Issues with visualize.plot()

aksvaidya, 2022-09-30:
I was able to successfully install pyslm and run the basic example code. However, I am not getting any output from `pyslm.visualise.plot(layer, plot3D=False, plotOrderLine=True)`

Running on windows Python 3.7.9 PythonSLM 0.5.0. Any suggestions on how to address this would be appreciated.

drlukeparry, 2022-09-30:
If you are not getting any output, it is likely there are two issues:

1) The data structure for Layers including Hatch and Contour Geometry is empty. Verify the output is correctly generated by using a debugging tool in the python environment if possible. 
2) Your python environment is not correctly setup to show a Matplotlib Figure window
      - after that function try (`import matplotlib.pyplot as plt; plt.show(block=True)` )

aksvaidya, 2022-09-30:
Thank you for such a quick reply. I got it to work using method 2. I will double check to make sure my environment is set up correctly. 

Thank you.

closed 2022-09-30 by aksvaidya