drlukeparry/pyslm #24: ClipperException: The path is invalid for clipping

MarcelHeu, 2022-10-10:
Hello,

[numpyArray.txt](https://github.com/drlukeparry/pyslm/files/9746469/numpyArray.txt)
I am using pyslm version = '0.4.2'

If i try to hatch a sliced contour (numpy array is attached within the *.txt file) with the following hatching settings i always got an ClipperException error:
 
```
Traceback (most recent call last):

  File "C:\Users\MarcelHeuser\AppData\Local\Temp\ipykernel_6728\2481062967.py", line 18, in <cell line: 18>
    layer = myHatcher.hatch([npArray])

  File "C:\Developer\venvTEST\lib\site-packages\pyslm\hatching\hatching.py", line 929, in hatch
    clippedPaths = self.clipLines(paths, hatches)

  File "C:\Developer\venvTEST\lib\site-packages\pyslm\hatching\hatching.py", line 432, in clipLines
    pc.AddPath(BaseHatcher.scaleToClipper(boundary), pyclipper.PT_CLIP, True)

  File "external\pyclipper\pyclipper.pyx", line 616, in pyclipper.Pyclipper.AddPath

ClipperException: The path is invalid for clipping
```

I have extracted this contour from a sliced geometry. You can use the code below to reproduce the error.

```
import numpy as np
from pyslm import hatching as hatching

# load numpy array
npArray = np.load("numpyArray.txt")

# generate Hatcher
myHatcher = hatching.Hatcher()

# Set the base hatching parameters
myHatcher.hatchAngle        = 66.7
myHatcher.volumeOffsetHatch = 0.08
myHatcher.spotCompensation  = 0.025
myHatcher.numInnerContours  = 0
myHatcher.numOuterContours  = 1
myHatcher.hatchDistance     = 0.12
myHatcher.hatchingEnabled   = True
myHatcher.hatchSortMethod   = hatching.AlternateSort()

# Hatch the geomSlice
layer = myHatcher.hatch([npArray])
```

Do you know what could be the reason for the error?

Best regards,
Marcel

drlukeparry, 2022-10-13:
Hi Marcel,

With version PySLM 0.5 dev, I didn't have any issues or an error was not reproducible. From the error message it is likely an invalid path is being provided. It is possible with the default contour offset results with an empty interior border. In this situation, the generates hatch infills cannot be clipped.

I remember resolving a similar issue, but cannot recall if this was included in the released PySLM v0.5.

If you can confirm with version 0.5 if the same issue occurs, that would be helpful.

MarcelHeu, 2022-10-14:
Hey Luke,

I tested it within a new virtual environment with version 0.5.0 but the identical error occurs:

```
  File "C:\Users\MarcelHeuser\Desktop\testpyslm\untitled2.py", line 28, in <module>
    layer = myHatcher.hatch([npArray])

  File "C:\Users\MarcelHeuser\Desktop\testpyslm\venvpyslm\lib\site-packages\pyslm\hatching\hatching.py", line 933, in hatch
    clippedPaths = self.clipLines(paths, hatches)

  File "C:\Users\MarcelHeuser\Desktop\testpyslm\venvpyslm\lib\site-packages\pyslm\hatching\hatching.py", line 433, in clipLines
    pc.AddPath(BaseHatcher.scaleToClipper(boundary), pyclipper.PT_CLIP, True)

  File "external\pyclipper\pyclipper.pyx", line 616, in pyclipper.Pyclipper.AddPath

ClipperException: The path is invalid for clipping
```

If i change the spotCompensation parameter to 0.026 or 0.024 it is working without any error, really strange.

drlukeparry, 2022-10-15:
Hi Marcel,

Thank you for taking the time to confirm this is also an issue in the current v0.5 release. 

I will confirm the behaviour separately and identify the difference between my development version. It is likely that the offsetting leaves no shapes, therefore, this error is incorrectly caught in the older released version. 

I will leave this open for the time-being.

closed 2024-02-20 by drlukeparry