drlukeparry/pyslm #34: .rdn format generation and reading build files

noedev86, 2023-02-05:
Hi, 
I wanted to know if there is any python example code to read and compile a .rdn file from Realizer, as well as the .rea file format. I only saw the .mtt format available.
Thank you

drlukeparry, 2023-02-06:
Hi,

The older Realizer .rea file format is supported in libSLM, but I have not fully tested this recently. This has been superceded by the modern version apparently adopted by DMGMori, however, I don't have access to any of these files. The .rdn file from what I understand is a similar construct ot the .rea format but with additional metadata.

noedev86, 2023-02-07:
and how did you do it for the .rea file? did the company give you some sort of library of any sort to help you compile it ? Because I'm working on this and it seems that you need a part of the code in order to compile it in their format.

drlukeparry, 2023-02-07:
The .rea file format specification or supporting library was unavailable and also was not provided by Realiser GmbH at the time. Therefore I had to reverse engineer the file format accordingly. As a result, access to the translator is only for research purposes because licensing agreement with the company is likely be required.

closed 2024-07-03 by drlukeparry