drlukeparry/pyslm #39: Improper array size error in identifysupportregions

Intech1, 2023-10-08:
While attempting to generate support material for a part, I receive the following error. It seems to be an array size mismatch, but I am unsure of the current cause.

[image]

drlukeparry, 2023-10-08:
Thank you for sharing the issue. 

This is known problem that can happen when some of the settings are not finally tuned.  Normally, the identified flattened  support region is processed into a polygon and a mesh is generated. Each point in this mesh is subsequently projected back onto the original mesh to generate the support block volumes.

However, If the simplification factor (`splineSimplicationFactor`) and the offset factors too large (`innerSupportEdgeGap`, `outerSupportEdgeGap`), these points may not coincidentally not project or intersect back with the mesh. 

I would suggest the following in this case:
- reduce the `rayProjectionResolution` to a lower value (e.g. 0.1 ) to capture the overhang boundaries regions more precisely
- reduce `splineSimplicationFactor` or set to 0 to eliminate over simplification of the boundary
- Adjust and increase the offset values from the support boundary for both `innerSupportEdgeGap`, and outerSupportEdgeGap

Intech1, 2023-10-09:
You're welcome!! 
Thank you for the suggestion! I took your advice into consideration and made some adjustments to the code based on your recommendations but facing the same issue.
I have attached the code file and a portion of the STL file for your reference. I would greatly appreciate it if you could kindly assist me in resolving this issue. Your expertise and guidance would be invaluable in helping me overcome this challenge.
Thank you once again for your support and willingness to assist.

[support_structure.zip](https://github.com/drlukeparry/pyslm/files/12843062/support_structure.zip)

drlukeparry, 2023-10-10:
Thank you for kindly sharing the example, which is very helpful to diagnose the issue further. 

I'll take a look and see what geometry related issue can be observed.

drlukeparry, 2023-10-11:
I looked over this in the development release (dev branch), which has further improvements and checks regarding this similar issue and found that it could generate the lattice with your support parameters. 

[image]

It could be that I introduced a method to catch some of errors related to this issue. Presumably you are using the version 0.5 of PySLM?

Intech1, 2023-10-13:
Hello, Good Day!!!
Thank you for your valuable insights and suggestions it helped us to understand the problem and its resolved the issue.

closed 2024-02-20 by drlukeparry