drlukeparry/pyslm #43: Trouble running example_support_structure.py on 0.6.0 (dev branch)

david-bue, 2024-03-19:
I am sitting on the dev branch, trying to set up the new support generation engine. I started with the example code and realized that the bracket stl file is not on the repo. I tried just with the nut.stl file instead, but I have a hard time getting the code to run. I keep getting this error:

Traceback (most recent call last):
  File "...\pyslm-dev\examples\example_support_structure.py", line 85, in <module>
    supportBlockRegions = supportGenerator.identifySupportRegions(myPart, OVERHANG_ANGLE, True)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "...\AppData\Local\Programs\Python\Python312\Lib\site-packages\pythonslm-0.6.0-py3.12.egg\pyslm\support\gridBlockSupport.py", line 1917, in identifySupportRegions
  File "...\AppData\Local\Programs\Python\Python312\Lib\site-packages\pythonslm-0.6.0-py3.12.egg\pyslm\support\support.py", line 1005, in identifySupportRegions
  File "...\AppData\Local\Programs\Python\Python312\Lib\site-packages\pythonslm-0.6.0-py3.12.egg\pyslm\support\geometry.py", line 347, in boolDiff
  File "...\AppData\Local\Programs\Python\Python312\Lib\site-packages\trimesh-4.2.0-py3.12.egg\trimesh\boolean.py", line 45, in difference
    raise ValueError("Not all meshes are volumes!")
ValueError: Not all meshes are volumes!

I can't figure out how to hunt down the issue, no matter how hard I try.

drlukeparry, 2024-03-19:
Hi David,

Firstly, please find attached the bracket.stl model used for testing. I cannot distribute this with the PySLM release because I am not sure which license (if any) exists for the model having lost the original source..
 
[bracket.stl.zip](https://github.com/drlukeparry/pyslm/files/14657789/bracket.stl.zip)

This will help for reference. However, it will work with other example models available but these need to be fully watertight for it to work with the manifold3d library.

I have looked over the specific error that is reported. This appears to be recently introduced in the new Trimesh release, when using the boolean operations. Trimesh now perfoms a pre-check to ensure the mesh has a volume and discards meshes that are empty. This happens when supports are not self-intersecting and connect to the base-plate. 

This also raised another observation that the support-volumes not connected to the base-plate are intersected with the original mesh.  

I will update the development release to fix this shortly, 

Luke

david-bue, 2024-03-20:
Thanks a bunch,
I will very much be looking forward to follow the updates.

drlukeparry, 2024-03-20:
Hi David,

I uploaded the changes yesterday night that should resolve this. Please feel free to try again and let me know if you have any problems with the support generation. 

There needs to be more detailed documentation with the support generation, but that is always challenging to fit in with other commitments. 

Luke

closed 2024-08-24 by drlukeparry