drlukeparry/pyslm #44: can you  parse slc files

zhenhuashao, 2024-07-03:
Now many people use slc files, for slc file parsing, want to seek advice

drlukeparry, 2024-07-03:
What are _.slc_ files-  are these full machine build files or slices, and which company/OEM use them on their system?

zhenhuashao, 2024-07-03:
This is the slicing data from magics slicing software.slc

zhenhuashao, 2024-07-03:
[a.zip](https://github.com/user-attachments/files/16086958/a.zip)
Compress the files in the package：
a.slc : Layer fill file
a_H.cli : Layer profile
a_s.slc  ：Supporting file

drlukeparry, 2024-07-03:
.CLI files are supported and can be provided as a python script. 

I don't fully understand the purpose of the _.slc_ file as it appears to be only the boundary/countours representing slices of a part geometry. These normally may be represented by internally by the .cli files so I don't fully understand what benefit you would have by being able to interpret this additional _.slc_ file?

closed 2024-07-13 by zhenhuashao