drlukeparry/pyslm #46: Bug in Support Generation for large and complex stl parts

Vipul-M-Rawal, 2024-08-21:
Hi, I am facing issues generating block supports for the given parts. I am getting almost the same problems for many parts. To solve the issues I made some changes in the parameters also, but still the issue persists.
I am attaching the stl file and the code below:

`"""
Support generation script - Shows how to generate basic block supports using PySLM
"""

import time
import numpy as np
import logging

from matplotlib import pyplot as plt
from pyslm.core import Part
import pyslm.support

import vispy
import trimesh
import trimesh.creation
import os
"""
Uncomment the line below to provide debug messages for OpenGL - if issues arise.
"""
# vispy.set_log_level('debug')

logging.getLogger().setLevel(logging.INFO)
vispy.use(app='pyqt5')  # or use another supported backend

os.environ['LIBGL_ALWAYS_SOFTWARE'] = '1'

## CONSTANTS ####
OVERHANG_ANGLE = 45 # deg - Overhang angle
# # Record the start time
start_time = time.time()
"""
Set the Geometry for the Example using a complicated topology optimised bracket geometry
"""
myPart = Part('myPart')
myPart.setGeometry("C:\\Desktop\\3-D Objects\\Heat Sink 3.stl", fixGeometry=True)
#myPart.scaleFactor = 4.0
myPart.rotation = [0.0, 0.0, 0.0] #[62.0, 50.0, -40.0] #[10, 70, 30] #[62.0, 50.0, -40.0]  #[-70.0, 50.0, -30.0] #[62.0, 50.0, -40.0]
#myPart.rotation = [76,35,-13] #[-25,0,5] #[76,35,-13]#[62,50,-40.0]

myPart.scaleFactor = 1.0
myPart.dropToPlatform(10)

""" Extract the overhang mesh - don't explicitly split the mesh"""
overhangMesh = pyslm.support.getOverhangMesh(myPart, OVERHANG_ANGLE,
                                             splitMesh=False, useConnectivity=True)

overhangMesh.visual.face_colors = [254.0, 0., 0., 254]

"""
Generate the geometry for the supports (Point and Edge Over Hangs)
"""
# First generate point and edge supports
# pointOverhangs = pyslm.support.BaseSupportGenerator.findOverhangPoints(myPart)
# overhangEdges  = pyslm.support.BaseSupportGenerator.findOverhangEdges(myPart)

"""
Generate block supports for the part.

The GridBlockSupportGenerator class is initialised and the parameters below are specified as a reasonable starting
defaults for the algorithm. The GridBlockSupport generator overrides the BlockSupportGenerator class and provides
additional methods for generating a grid-truss structure from the support volume.
"""
supportGenerator = pyslm.support.GridBlockSupportGenerator()
supportGenerator.rayProjectionResolution = 0.05 # [mm] - The resolution of the grid used for the ray projection
# supportGenerator.rayProjectionResolution = 0.05 # [mm] - The resolution of the grid used for the ray projection
supportGenerator.innerSupportEdgeGap = 0.3      # [mm] - Inner support offset used between adjacent support distances
supportGenerator.outerSupportEdgeGap = 0.3      # [mm] - Outer support offset used for the boundaries of overhang regions
[... 254 lines cut ...]
    canvas.context.flush_commands()
  File "C:\Users\vipul.k\AppData\Roaming\Python\Python312\site-packages\vispy\gloo\context.py", line 172, in flush_commands
    self.glir.flush(self.shared.parser)
  File "C:\Users\vipul.k\AppData\Roaming\Python\Python312\site-packages\vispy\gloo\glir.py", line 584, in flush
    self._shared.flush(parser)
  File "C:\Users\vipul.k\AppData\Roaming\Python\Python312\site-packages\vispy\gloo\glir.py", line 506, in flush
    parser.parse(self._filter(self.clear(), parser))
  File "C:\Users\vipul.k\AppData\Roaming\Python\Python312\site-packages\vispy\gloo\glir.py", line 824, in parse
    self._parse(command)
  File "C:\Users\vipul.k\AppData\Roaming\Python\Python312\site-packages\vispy\gloo\glir.py", line 800, in _parse
    ob.set_framebuffer(*args)
  File "C:\Users\vipul.k\AppData\Roaming\Python\Python312\site-packages\vispy\gloo\glir.py", line 1758, in set_framebuffer
    self._validate()
  File "C:\Users\vipul.k\AppData\Roaming\Python\Python312\site-packages\vispy\gloo\glir.py", line 1823, in _validate
    raise RuntimeError(_bad_map.get(res, 'Unknown framebuffer error: %r.'
RuntimeError: Target not equal to GL_FRAMEBUFFER
ERROR: Invoking <bound method Canvas.on_draw of <Canvas (PyQt5) at 0x22ccc1ee4e0>> for DrawEvent
ERROR:vispy:Invoking <bound method Canvas.on_draw of <Canvas (PyQt5) at 0x22ccc1ee4e0>> for DrawEvent

INFO:root:Time taken for support geometry creation: 46.22 seconds
C:\Users\vipul.k\AppData\Roaming\Python\Python312\site-packages\pyglet\libs\win32\__init__.py:318: UserWarning: Could not set COM MTA mode. Unexpected behavior may occur.
  warnings.warn("Could not set COM MTA mode. Unexpected behavior may occur.")
  
  
[Structure-pack_200 mm.zip](https://github.com/user-attachments/files/16690801/Structure-pack_200.mm.zip)

drlukeparry, 2024-08-23:
the error appears related to the vispy module - it is using the Pyglet backend which I have observed to create some problems in the past depending on the platform used. Additionally, the functionality depends on having reasonably modern openGL graphics drivers that support floating-point depth framebuffers, otherwise this method breaksdown. I imagine under certain virtualised platforms this could be problematic. 

 I would suggest to install PyQt5 / PyQt6 and set the corresponding backend for vispy to use this and see if this fixes the problem you are encountering.

QuentinoPersonal66, 2025-01-09:
Hi Luke,
I faced the same problem even though I have PyQt5 and vispy installed.
versions are
vispy                     0.14.3
pyqt5                     5.15.11

I am using windows 11. 
which version are you using ?

drlukeparry, 2025-01-10:
@QuentinoPersonal66 - can you provide any information on your graphics card (+driver) - my suspicion is that some Graphics Cards don't support floating point framebuffers and is why the error is raised by Vispy. 

This happened to me previously on an older system (Ubuntu OS with integrated Intel GPU), which only provided RGB8 format and created a variety of issues obtaining the projection map.

QuentinoPersonal66, 2025-01-10:
@drlukeparry my graphic card is intel(R) graphics. 
driver version is 32.0.101.5972.

QuentinoPersonal66, 2025-01-10:
I found out that decreasing the rayprojection resolution can help

drlukeparry, 2025-01-10:
@QuentinoPersonal66 - that's helpful, thank you for reporting that. Likely the issues is the framebuffer size is being exceeded at the ray projection resolution chosen for a large support area. Some GPUs and Graphics Drivers have a restriction of the framebuffer size (in pixels). The default ray projection resolution is a tradeoff between accuracy and detail for parts with not a large overhang size.

closed 2026-02-14 by drlukeparry