drlukeparry/pyslm #51: Offset Support Volume Generated with STL File.

Vipul-M-Rawal, 2024-12-17:
When generating the support volume for an STL file using PySLM, the resulting support volume appears to be offset from the expected position. This misalignment makes the generated support geometry unusable for further operations like slicing or additive manufacturing.

**Expected Behavior**
The support volume should align perfectly with the base STL geometry.

**Actual Behavior**
The generated support volume is offset relative to the input STL file.

**Parameters I am using:**

supportGenerator.innerSupportEdgeGap = 0.3      # [mm] - Inner support offset used between adjacent support distances
supportGenerator.outerSupportEdgeGap = 2.0     # [mm] - Outer support offset used for the boundaries of overhang regions
supportGenerator.simplifyPolygonFactor = 0.5    #      - Factor used for simplifying the overall support shape
supportGenerator.triangulationSpacing = 2.0     # [mm] - Used for triangulating the extruded polygon for the bloc
supportGenerator.minimumAreaThreshold = 0.2     # Minimum area threshold to not process support region'
supportGenerator.triangulationSpacing = 4       # [mm^2] - Internal parameter used for generating the mesh of the volume
supportGenerator.supportBorderDistance = 1.0    # [mm]
supportGenerator.numSkinMeshSubdivideIterations = 2

# Support teeth parameters
supportGenerator.useUpperSupportTeeth = True
supportGenerator.useLowerSupportTeeth = True
supportGenerator.supportWallThickness = 1.0         # [mm] - The thickness of the upper and support walls to strengthen teeth regions
supportGenerator.supportTeethTopLength = 0.1        # [mm] - The length of the tab for the support teeth
supportGenerator.supportTeethHeight = 1.5           # [mm] - Length of the support teeth
supportGenerator.supportTeethBaseInterval = 1.5     # [mm] - The interval between the support teeth
supportGenerator.supportTeethUpperPenetration = 0.2 # [mm] - The penetration of the support teeth into the part

supportGenerator.splineSimplificationFactor = 0.01 # - Specify the smoothing factor using spline interpolation for the support boundaries
supportGenerator.gridSpacing = [5,5] # [mm] The Grid

[image]

drlukeparry, 2024-12-17:
I cannot understand or resolve issue because I cannot determine the orientation of the supports from the attached picture.

Please provide a full example with the model to verify the issue, including the versions of the libraries (including PySLM) that have been used in order to reproduce this.

Vipul-M-Rawal, 2024-12-18:
I am using the given code. I have attached the model stl file along with the image :
`"""
Support generation script - Shows how to generate basic block supports using PySLM
"""

import numpy as np
import logging

from matplotlib import pyplot as plt
from pyslm.core import Part
import pyslm.support

import vispy
import trimesh
import trimesh.creation

"""
Uncomment the line below to provide debug messages for OpenGL - if issues arise.
"""
# vispy.set_log_level('debug')

logging.getLogger().setLevel(logging.INFO)

## CONSTANTS ####
OVERHANG_ANGLE = 55 # deg - Overhang angle

"""
Set the Geometry for the Example using a complicated topology optimised bracket geometry
"""
myPart = Part('myPart')
myPart.setGeometry("C:\\Desktop\\3-D Objects\\Step.stl", fixGeometry=True)
#myPart.scaleFactor = 4.0
myPart.rotation = [0.0, 0.0, 0.0] #[62.0, 50.0, -40.0] #[10, 70, 30] #[62.0, 50.0, -40.0]  #[-70.0, 50.0, -30.0] #[62.0, 50.0, -40.0]
#myPart.rotation = [76,35,-13] #[-25,0,5] #[76,35,-13]#[62,50,-40.0]

myPart.scaleFactor = 1.0
myPart.dropToPlatform(10)

""" Extract the overhang mesh - don't explicitly split the mesh"""
overhangMesh = pyslm.support.getOverhangMesh(myPart, OVERHANG_ANGLE,
                                             splitMesh=False, useConnectivity=True)

overhangMesh.visual.face_colors = [254.0, 0., 0., 254]

"""
Generate the geometry for the supports (Point and Edge Over Hangs)
"""
# First generate point and edge supports
pointOverhangs = pyslm.support.BaseSupportGenerator.findOverhangPoints(myPart)
overhangEdges  = pyslm.support.BaseSupportGenerator.findOverhangEdges(myPart)

"""
Generate block supports for the part.

The GridBlockSupportGenerator class is initialised and the parameters below are specified as a reasonable starting
defaults for the algorithm. The GridBlockSupport generator overrides the BlockSupportGenerator class and provides
additional methods for generating a grid-truss structure from the support volume.
"""
supportGenerator = pyslm.support.GridBlockSupportGenerator()
supportGenerator.rayProjectionResolution = 0.05 # [mm] - The resolution of the grid used for the ray projection
supportGenerator.innerSupportEdgeGap = 0.3      # [mm] - Inner support offset used between adjacent support distances
supportGenerator.outerSupportEdgeGap = 0.3      # [mm] - Outer support offset used for the boundaries of overhang regions
supportGenerator.simplifyPolygonFactor = 0.5    #      - Factor used for simplifying the overall support shape
supportGenerator.triangulationSpacing = 2.0     # [mm] - Used for triangulating the extruded polygon for the bloc
supportGenerator.minimumAreaThreshold = 0.1     # Minimum area threshold to not process support region'
supportGenerator.triangulationSpacing = 4       # [mm^2] - Internal parameter used for generating the mesh of the volume
supportGenerator.supportBorderDistance = 1.0    # [mm]
supportGenerator.numSkinMeshSubdivideIterations = 2

# Support teeth parameters
[... 66 lines cut ...]

""" Identify the sides of the block extrudes """
s1 = trimesh.Scene([myPart.geometry] + blockSupports)

"""
The following section exports the group of support structures from the trimesh scene. 
"""

with open('overhangSupport.glb', 'wb') as f:
    f.write(trimesh.exchange.gltf.export_glb(s1, include_normals=True))

"""
Show only the volume block supports generated
"""
DISPLAY_BLOCK_VOLUME = True

if DISPLAY_BLOCK_VOLUME:
    s2 = trimesh.Scene([myPart.geometry, overhangMesh,
                        blockSupports])
    s2.show()
`

[image]

[Step.zip](https://github.com/user-attachments/files/18175677/Step.zip)

Vipul-M-Rawal, 2025-03-09:
Hi Luke, 
I tweaked some parameters, and now the offset and curve on corner problem has been solved, but now it is not generating supports for some parts that worked earlier. I am using the following parameters :
`volume_generator = SupportVolumeGenerator()
volume_generator.overhangAngle = 45.0
volume_generator.rayProjectionResolution = 0.05 # [mm] - The resolution of the grid used for the ray projection
volume_generator.innerSupportEdgeGap = 0.0      # [mm] - Inner support offset used between adjacent support distances
volume_generator.outerSupportEdgeGap = 0.0      # [mm] - Outer support offset used for the boundaries of overhang regions
volume_generator.simplifyPolygonFactor = 0.5    #      - Factor used for simplifying the overall support shape
volume_generator.triangulationSpacing = 2.0     # [mm] - Used for triangulating the extruded polygon for the bloc
volume_generator.minimumAreaThreshold = 0.1     # Minimum area threshold to not process support region'
volume_generator.triangulationSpacing = 1.0       # [mm^2] - Internal parameter used for generating the mesh of the volume
volume_generator.supportBorderDistance = 0.0    # [mm]
volume_generator.numSkinMeshSubdivideIterations = 2

# Support teeth parameters
volume_generator.useUpperSupportTeeth = True
volume_generator.useLowerSupportTeeth = True
volume_generator.supportWallThickness = 0.0         # [mm] - The thickness of the upper and support walls to strengthen teeth regions
volume_generator.supportTeethTopLength = 0.1        # [mm] - The length of the tab for the support teeth
volume_generator.supportTeethHeight = 1.5           # [mm] - Length of the support teeth
volume_generator.supportTeethBaseInterval = 1.5     # [mm] - The interval between the support teeth
volume_generator.supportTeethUpperPenetration = 0.2 # [mm] - The penetration of the support teeth into the part

volume_generator.splineSimplificationFactor = 0.0                                                                            # - Specify the smoothing factor using spline interpolation for the support boundaries
volume_generator.gridSpacing = [5,5] # [mm] The Grid`

And I am getting the following error :

Traceback (most recent call last):
  File "C:\Users\vipul.k\OneDrive - Intech Additive Solutions Pvt Ltd\Pictures\Original_Memory_SupportGen_Pyslm\pyslm\SupportVolume.py", line 171, in <module>
    support_regions = volume_generator.identifySupportRegions(myPart, volume_generator.overhangAngle, subregions=subregions)
  File "C:\Users\vipul.k\OneDrive - Intech Additive Solutions Pvt Ltd\Pictures\Original_Memory_SupportGen_Pyslm\pyslm\pyslm\support\SupportVolumeGenerator.py", line 322, in identifySupportRegions
    blockSupportMesh = boolDiff(extrudedBlock, cutMesh)
  File "C:\Users\vipul.k\OneDrive - Intech Additive Solutions Pvt Ltd\Pictures\Original_Memory_SupportGen_Pyslm\pyslm\pyslm\support\geometry.py", line 347, in boolDiff
    outMesh = trimesh.boolean.difference([meshA, meshB], engine='manifold', check_volume=False)
  File "C:\Users\vipul.k\AppData\Local\Programs\Python\Python39\lib\site-packages\trimesh\boolean.py", line 49, in difference
    return _engines[engine](meshes, operation="difference", **kwargs)
  File "C:\Users\vipul.k\AppData\Local\Programs\Python\Python39\lib\site-packages\trimesh\boolean.py", line 134, in boolean_manifold
    raise ValueError("Not all meshes are volumes!")
ValueError: Not all meshes are volumes!

drlukeparry, 2026-02-14:
Will be resolved as part of the new occlusion based block support generation in 0.7.0

closed 2026-02-14 by drlukeparry