drlukeparry/pyslm #57: error in Support generation

QuentinoPersonal66, 2025-01-13:
Hi,

I tried the library on various stl files and I encountered this error several time 

'''
Traceback (most recent call last):
  File "C:\Users\E129000\OneDrive - RMIT University\Documents\Codes\Python\PYSLM_EXAMPLES\example_support_structure.py", line 93, in <module>
    supportBlockRegions = supportGenerator.identifySupportRegions(myPart, OVERHANG_ANGLE, True)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\E129000\AppData\Local\anaconda3\envs\PYSLM\Lib\site-packages\pyslm\support\gridBlockSupport.py", line 1936, in identifySupportRegions
    supportBlocks = super().identifySupportRegions(part, overhangAngle, findSelfIntersectingSupport)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\E129000\AppData\Local\anaconda3\envs\PYSLM\Lib\site-packages\pyslm\support\support.py", line 1014, in identifySupportRegions
    blockSupportMesh = boolDiff(extrudedBlock, cutMesh)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\E129000\AppData\Local\anaconda3\envs\PYSLM\Lib\site-packages\pyslm\support\geometry.py", line 347, in boolDiff
    outMesh = trimesh.boolean.difference([meshA, meshB], engine='manifold', check_volume=False)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\E129000\AppData\Local\anaconda3\envs\PYSLM\Lib\site-packages\trimesh\boolean.py", line 49, in difference
    return _engines[engine](meshes, operation="difference", **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\E129000\AppData\Local\anaconda3\envs\PYSLM\Lib\site-packages\trimesh\boolean.py", line 134, in boolean_manifold
    raise ValueError("Not all meshes are volumes!")
ValueError: Not all meshes are volumes!
'''

Sometimes it happens when a plane surface is exactly located at the baseplate, but it also happens in other situation that I struggle to understand. 
Any explanation on this error would be very helpful

drlukeparry, 2025-01-13:
The support generation is not entirely full proof and is dependent on the complexity of the part geometry provided. 

The robustness of the current approach relies on extruding the overhang surface identified during the support  generation, and then calculating the boolean difference with the original part using the manifold library. If the overhang surface has noisy   or protruding surfaces these can under some situations become non-manifold/watertight. Alternatively, very large overhang areas with many corresponding areas tends to be problematic from my experience. This corresponds with line:

```
File "C:\Users\E129000\AppData\Local\anaconda3\envs\PYSLM\Lib\site-packages\pyslm\support\support.py", line 1014, in identifySupportRegions

blockSupportMesh = boolDiff(extrudedBlock, cutMesh)
```

I think there are better ways now to address this in the future, because the ray projection technique now exists compared to when the algorithm was originally designed. However, it is tricky without having lower level access to a boolean CSG library to perform self-intersection test

Secondly the block support is extended currently to the base-plate so if the part is in near contact it can be problematic and is best to avoid the overhang surfaces being located precisely at Z=0. I may look further into this specific issue.  

If you are able to share screenshots of the parts + orientation that may help provide further diagnosis of the cause.

QuentinoPersonal66, 2025-01-28:
Attached are two screenshots of a part for which I could not generate supports successfully. The rotated version works though. Orientation is Z pointing up in the two picture

[image]

[image]