drlukeparry/pyslm #61: Convert to C++ lib

loadship, 2025-06-24:
"Excellent work! Have you considered porting this to C++? I couldn't find any discussion opened on that, so I have to reach out to you here — although I realize this might not be the most appropriate place."

drlukeparry, 2025-06-24:
Thank you for your kind feedback.

Most of the low-level operations (i.e. polygon clipping) are already performed in c++ libraries such as Shapely and ClipperLib2. The current bottleneck is related to the mesh operations in Trimesh - which have a reliance on numpy and use of double (float64) numerical representations that are not as performant. There are ofcourse overheads using python, however, the purposes of PySLM is to remain adaptable and usable for research purposes. 

Some of the underlying algorithms are relatively poor (hatching, slicing). There can be opportunities for GPU computation, but the current purposes would not provide substantial benefits without more significant commercial usage. 

Rust would be an ideal candidate to take this forward. The author of trimesh has create a project rmesh as an equivalent high performance port to the original trimesh library.

loadship, 2025-06-25:
Got it, thank you for your answer. CuraEngine is a good option for slicing, but without any benefit or incentive, it's hard to stay motivated to push this work forward.

loadship, 2025-06-25:
I would like to seek your guidance. From my preliminary review of your hatch algorithms—zigzag, stripe, and island strategies—it appears that region partitioning is not addressed. Could you please clarify if I have overlooked this aspect? It is important to note that modern SLM equipment typically employs multi-laser printing.

loadship, 2025-06-25:
I’ve now come up with a solution inspired by the concepts of zigzag, stripe, and island-based partitioning. By slightly offsetting the positions before clipping, I can simplify the clipping process. My approach is to minimize the contour units to reduce the computational cost of union operations. Thank you again — your insights were a great inspiration.