drlukeparry/pyslm #62: Bug using support generation

QuentinoPersonal66, 2025-07-07:
Hi The following Code 

"
import time
import pyslm
import numpy as np

import pyslm.support

from MAIN import file

SF = 1.0
ROT = np.array([1,0,0])
OVERHANG_ANGLE = 45

supportGenerator = pyslm.support.GridBlockSupportGenerator()
supportGenerator.rayProjectionResolution = 0.05 # [mm] - The resolution of the grid used for the ray projection
supportGenerator.innerSupportEdgeGap = 0.3      # [mm] - Inner support offset used between adjacent support distances
supportGenerator.outerSupportEdgeGap = 0.3      # [mm] - Outer support offset used for the boundaries of overhang regions
supportGenerator.simplifyPolygonFactor = 0.5    #      - Factor used for simplifying the overall support shape
supportGenerator.triangulationSpacing = 2.0     # [mm] - Used for triangulating the extruded polygon for the bloc
supportGenerator.minimumAreaThreshold = 0.1     # Minimum area threshold to not process support region'
supportGenerator.triangulationSpacing = 4       # [mm^2] - Internal parameter used for generating the mesh of the volume
supportGenerator.supportBorderDistance = 1.0    # [mm]
supportGenerator.numSkinMeshSubdivideIterations = 2

# Support teeth parameters
supportGenerator.useUpperSupportTeeth = True
supportGenerator.useLowerSupportTeeth = True
supportGenerator.supportWallThickness = 1.0         # [mm] - The thickness of the upper and support walls to strengthen teeth regions
supportGenerator.supportTeethTopLength = 0.1        # [mm] - The length of the tab for the support teeth
supportGenerator.supportTeethHeight = 1.5           # [mm] - Length of the support teeth
supportGenerator.supportTeethBaseInterval = 1.5     # [mm] - The interval between the support teeth
supportGenerator.supportTeethUpperPenetration = 0.2 # [mm] - The penetration of the support teeth into the part

supportGenerator.splineSimplificationFactor = 10 # - Specify the smoothing factor using spline interpolation for the support boundaries
supportGenerator.gridSpacing = [5,5] # [mm] The Grid

solidPart = pyslm.Part('PART')
solidPart.setGeometry(file)
solidPart.origin = [0.0, 0.0, 0.0]
solidPart.rotation = ROT
solidPart.scaleFactor = SF
solidPart.dropToPlatform()
Zmax = solidPart.boundingBox[-1]-solidPart.boundingBox[2]

supportBlockRegions = supportGenerator.identifySupportRegions(solidPart, OVERHANG_ANGLE, True)
for block in supportBlockRegions:
    block.trussWidth = 1.0

blockSupports = [block.supportVolume for block in supportBlockRegions]

Vs = 0 
for bs in blockSupports:
    if (bs.is_watertight ==True):
        Vs = Vs + bs.volume
print("estimated support volume by PYSLM is " + str(Vs) + " mm3" )
"

yield the following error

"
event physical size 949 825
Traceback (most recent call last):
  File "/home/ec2-user/Documents/PYTHON/ChoixProcess/PROCESS_ENVIRONMENTAL_PB.py", line 227, in <module>
    supportBlockRegions = supportGenerator.identifySupportRegions(solidPart, OVERHANG_ANGLE, True)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ec2-user/anaconda3/envs/PYSLM/lib/python3.12/site-packages/pyslm/support/gridBlockSupport.py", line 1936, in identifySupportRegions
    supportBlocks = super().identifySupportRegions(part, overhangAngle, findSelfIntersectingSupport)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ec2-user/anaconda3/envs/PYSLM/lib/python3.12/site-packages/pyslm/support/support.py", line 906, in identifySupportRegions
    raise Exception('Multi-polygons - error please submit a bug report')
Exception: Multi-polygons - error please submit a bug report
"

I will attache the corresponding file, but any idea on how to solve this ?

QuentinoPersonal66, 2025-07-07:
[JEB_066_INIT.zip](https://github.com/user-attachments/files/21099927/JEB_066_INIT.zip)

drlukeparry, 2025-07-07:
Further too add. I will incorporate a small fix to resolve the issue of support generation with regions located near the base-plate.

QuentinoPersonal66, 2025-07-08:
Hi Luke. Thank you for you feedback. It did solved the problem for this case. I do have issue with other geometries but I imagine I should play a bit with the parameters on a case-by-case basis.